ivanseidel/DueTimer #35: What happend if Callback take longer...

xxseabxx, 2015-01-18:
Hello!
First of all thx for the great library it safes me a lot of time.

What happend if the callback function take longer than the frequent is set to?
For instance: I set the frequence of the timer to 3HZ. In my callback I wait for something (while readingPin is HIGH). Therefore what happens if the callback code take longer than 1/3 second? 
When I tried to simulate such situation  all instanced  Timers don't work anymore.

Here is an example
//start countdown with independed timer 
Timer0.attachInterrupt(countdownHandler).setFrequency(1).start(); 
Timer1 //other timer for 
....

void sequenceHandler()
{
digitalWrite(pinEOC, HIGH); //start loading 

while (digitalRead(pinEOC) != HIGH) 
{ //waiting for end of charge (normally 10microseconds) -->PROBLEM
}
... //do other stuff
}
}

My current solution is to monitor the time and interrupt it if it take longer than the frequency but it's not pretty :(

ivanseidel, 2015-01-19:
Try this and let me know if works:
```c++
void sequenceHandler(){
    // Disable Interrupts
    noInterrupts();

    // Do something that may take longer...
    while(something());

    // Enable Interrupts again and exit
    interrupts();
}
```

If that doesn't work let me know @xxseabxx

l-mb, 2015-01-19:
I thought interrupts are automatically disabled in an ISR; so no need to explicitly call noInterrupts(). If your ISR takes longer, new interrupts are queued, and will be delivered afterward (but not multiple times for the same IRQ). Of course, then your timing is screwed up.

I think the best way for what you want would be to find an alternative to the busy wait in the ISR. Since you're waiting for a HIGH->LOW transition on a digital PIN, you can handle that via an interrupt too:

attachInterrupt(pinEOC, handler, FALLING);

That'll probably need some reshuffling of your code to be more event-driven.

xxseabxx, 2015-01-20:
Thanks for the fast reply.
I will try it out in the next few days!

@l-mb I'm not sure how to change my code to work with the attachInterrupt - but i'm also the opinion it's better to work outside the sequence

I let you know my experience
Thanks

ivanseidel, 2015-06-21:
@xxseabxx 
Any news? I want to close this if possible

xxseabxx, 2015-07-05:
Hello 
I handle this with the 
attachInterrupt(pinEOC, handler, FALLING);

thanks for you reply and help.
regards sebastian

closed 2015-07-05 by xxseabxx