ivanseidel/DueTimer #38: Is it possible to attach external clocks to the timers

sriranjanr, 2015-06-21:
You mention abou external clock but is it possible to attach external clocks to your timer library?Please let me know how.

ivanseidel, 2015-06-21:
Hello @sriranjanr,
Right now the DueTimer only supports internal clock source, unless you change a few registers that I don't know...
You might see more useful information in the Datasheet.

Best,

closed 2015-06-21 by ivanseidel

sriranjanr, 2015-06-21:
TC_BMR register is there which needs to be altered to get the External clock input. If you can guide me I will add code to the library.

ivanseidel, 2015-06-21:
Sorry that I can't help you with this,
If it was a bug or something I could help you more, but this will require a
lot of deep-reading through the Datasheet, and I'm pretty busy right now...

I suggest you to seek someone that already implemented it on the Web, and
you can expand your search by looking for "ARM Cortex M3" instead of
"Arduino DUE".

But if you find the answer, please post here for others.

Best,

sriranjanr, 2015-06-21:
ok

ivanseidel, 2015-06-21:
I'm sorry that I couldn't help you, my knowledge on this matter was just enough to build this library

Best