ivanseidel/DueTimer #43: questions about DueTimer library

ApplefaceLisa, 2015-09-11:
Hi, I wanna use this library on my new arduino due board. I'm not sure on which arduino IDE version can this library be used? (Now I'm using 1.6.2 and it do not support due board.)

The second question is, if I use timer.stop(), then use timer.start(), will the timer count from zero again, or will it continue to count from where it was stopped? For example, if I set the period to 500ms, and I stop it at 200ms, then start it again, will it send interrupt after 300ms? Because in Timer1 library used for mega board, if I use timer.start(), it continues counting; and if I use timer.restart(), it count from zero.

The third question is, do all the 8 timers have the same resolution? What's the minimum period can I set? And what's the max number before they overflow?

Thank you.

ivanseidel, 2015-09-11:
Hello, If I'm not mistaken, the Due Borard has to be installed in the devices manager on the new versions of the IDE.

Also, the start restart the time. This has to be confirmed (and is a nice thing to be on the library Docs).

Its has been a while that I don't change things on Due Timer. if you can help improve, you will be more than welcome.

ApplefaceLisa, 2015-09-11:
Hi Ivan, thank you for reply. I just tested my code using your lib and it works. And I also found the way to install due board. Till now everything goes fine. I would like to help improve, but I am brand new to Arduino, so right now I am not able to do it. But hope I can in near future.:)

ivanseidel, 2015-09-12:
Great!
Don't take too long to learn Git. It's awesomeness is just showed when you learn it. It will engage you to a large community of open-source projects. It's a "must" for any today's maker

closed 2015-09-12 by ivanseidel