ivanseidel/DueTimer #45: Bug with callback period?

SquidFigther, 2016-02-13:
Hi, I am using Arduino DueTimer in a school project but is having a problem. I use the interrupt created from a timer, however myHandler does have a lot of operations and this produce a bug I think. The bug is that the interrupt callback is called with a period time less then the given. I send an example when the callback function does have a delay in it to demonstrate the error. Note that the delay is much smaller then the interrupt period.

Would be realy grateful if you could look at this bug or explain for me why its not working as I think it would.

link:
https://drive.google.com/file/d/0B4hLj0ggOnuZaC1XR2tfcmV1WkU/view?usp=sharing

Best regards,
Alexander Andersson

ivanseidel, 2016-02-22:
solved in email =)

closed 2016-02-22 by ivanseidel

kapyaar, 2016-03-11:
Hello,

I have this same issue. I just downloaded the master and was testing it. Wondering if you already updated the master?

ivanseidel, 2016-03-11:
You cannot use `delay()` inside interrupt handlers, because the delay function is basically a timer counter. When you `wait` inside a interrupt handler, you are basically waiting for the delay counter to increase, however, that won't happen because you are already `in` the interrupt.

To solve that, use `delayMicroseconds(...)` instead of `delay(...)`