ivanseidel/DueTimer #47: "#ifdef __arm__" of DueTimer.h  not working (anymore)?

BluefireXD, 2016-04-03:
Hello. In my Projekt the DueTimer.h is now working because somehow the #ifdef __arm__ is not found. 
After a bit of search i found out that in my sam.h which is included from in chip.h (and chip.h is included in Arduino.h) is no __arm__ defined. Mybe they changed it in the current version a bit.. idk. So in sam.h is written:

#define part_is_defined(part) (defined(__ ## part ## __)) ......
#define SAM3X8 ( \
    part_is_defined( SAM3X8C ) || \
    part_is_defined( SAM3X8E ) || \
    part_is_defined( SAM3X8H ) )

So after i changed the "#ifdef __arm__"  to  "#if defined (__arm__) || defined (__SAM3X8E__)" in your DueTimer.h it works fine again.
Maybe you should have a look on this issue^^.

By the way... Thanks a lot for your work! Its a realy great Lib :)

ivanseidel, 2016-04-21:
Hi @BluefireXD! Did you manage to solve this? We must guarantee that it will work across everyone using it..

BluefireXD, 2016-04-21:
Hi, Yes in my case the solution works fine. 
Thjs Line: #if defined (__arm__) || defined (__SAM3X8E__) should work for both cases where only __arm__ is defined and where only __SAM3X8E__ is defined.

ivanseidel, 2018-04-30:
this is fixed by latest merges. We are now checking for `__SAM3X8E__`

closed 2018-04-30 by ivanseidel