ivanseidel/DueTimer #49: This library doesn't work

cyrilfr, 2016-09-05:
This library cannot be installed:
Invalid library found in /root/Arduino/libraries/DueTimer: Missing 'core-dependencies' from library

Daniel-dk, 2016-09-05:
did you follow the installation instructions on : https://github.com/ivanseidel/DueTimer ?
and do you have "Arduino SAM Boards"  installed as a board in the Arduino IDE ( via boards manager ? ) 
Tools -> Board -> Boards Manager -> Arduino SAM boards

cyrilfr, 2016-09-05:
No, I'm working with Arduino Due.

closed 2016-09-05 by cyrilfr

Daniel-dk, 2016-09-05:
Arduino SAM Boards is the installation package that contains Arduino Due ( cortex M3 ) core dependancies

cyrilfr, 2016-09-05:
I'm using a patched (bossac) version of Arduino IDE 1.5.4 compatible with the SAM3x of the UDOO Quad board. I must select "Arduino Due (Native Port)" in the "Board" menu. I don't have a SAM Boards entry.