ivanseidel/DueTimer #59: Blocking Interrupt

thalesmaoa, 2017-06-09:
Hi, I'm using DAC to generate a low frequency sine with a handler.
I'm also using another interrupt to update LCD. 

The problem is that, when LCD calls, the DAC stop cycling and stay constant.
How can I build a blocking interrupt?

```
void atualizaLCD(){
    //lcd.clear();
    lcd.setCursor(0,0); //Start at character 0 on line 0
    lcd.print(strOnda[wave0]);
    lcd.setCursor(0,1);
    lcd.print(strOnda[wave1]);

    // Imprime amplitude com uma casa
    lcd.setCursor(4,1);
    analogSync = analogRead(A6);
    angleSync = analogSync*180.0/1023.0;
    angleSync_i = analogSync*maxSamplesNum/511.5;
    lcd.print( angleSync, 0 );
}

void ondaADC(){
  Timer1.stop();
  i_wav++;
  if(i_wav == maxSamplesNum)  // Reset the counter to repeat the wave
    i_wav = 0;
  analogWrite(DAC0, waveformsTable[wave0][i_wav]);  // write the selected waveform on DAC0
  analogWrite(DAC1, waveformsTable[wave1][i_wav]);  // write the selected waveform on DAC1
  Timer1.start();
}

void setup(){
    analogWriteResolution(12);  // set the analog output resolution to 12 bit (4096 levels)
    Timer0.attachInterrupt(ondaADC).setFrequency((Freq_wav * 120)).start(); // 100 * 120 points = 1200Hz
    Timer1.attachInterrupt(atualizaLCD).setFrequency(1).start();
}
```

piflixe, 2017-06-10:
Hi there! 
Since updating lcd does not need real time, I would suggest updating the data that needs to be displayed in ISR but update the display in the normal main loop let's say every 300 ms or so.
This approach worked for me very well. 
Cheers, Felix

thalesmaoa, 2017-06-10:
This is definitely a good solution, but it doesn't fix my issue. 

Indeed this was only an exempl

ivanseidel, 2018-04-30:
Try adding `noInterrupts()` on the first line of `atualizaLCD()` and `interrupts()` on the last line. It will prevent interrupts from being called during that period

closed 2018-04-30 by ivanseidel