ivanseidel/DueTimer #61: precise timing

peiyzhou, 2018-02-13:
Hello,

Can anybody give me some tips to use this library for precise timing? like the arduino micros() function? I want to add accurate timestamps to the sensor outputs...

Thanks.

piflixe, 2018-02-14:
Hi Peiyzhou,

I used this library for my master thesis. You can find the code here:
https://github.com/piflixe/ILR/tree/master/ILR

The main file is ILR.ino which also include the DueTimer library. In setup.ino I initialise the timer
```
Timer3.setFrequency(sampleFreq);
Timer3.attachInterrupt(changeIndex).stop(); // init Timer in stop condition
```
which then calls the changeIndex.ino routine that does something in a very precise timing intervall.

Hope this helps?

Cheers Felix

peiyzhou, 2018-02-18:
Thank you! I am trying...