ivanseidel/DueTimer #64: Calling setFrequency() with timer active causes callback to stop running

kjfergu, 2018-05-09:
I have a program where I want to adjust a frequency of the timer on the fly with the output active. Essentially, I am generating a sine wave and need to sweep through a range of frequencies.

An example code would be:
```
Timer0.setFrequency(someFreq);
Timer0.start()
// Output looks good
Timer0.setFrequency(someOtherFreq);
// Output stops
Timer0.stop();
Timer0.start();
// Output looks good with correct new frequency
```

As a general rule, someFreq and someOtherFreq are close to each other (Within 1 hz), and both of them are in the 10-50 Hz range.

I'm not sure if changing frequency with the timer active is just unsupported, but I imagine it should be possible since all that is changing is the overflow counter as I understand it. It's possible that I'm doing something boneheaded as well.

henriksod, 2021-06-26:
I also need help with this. I am trying to control the speed of stepper motors by using setFrequency on the fly.

fergu, 2021-06-26:
@henriksod I ended up doing this by stopping the timer, adjusting the frequency, and restarting the timer like I showed above. It seems to work just fine.

Closing this as that's pretty much the solution I found that worked best.

closed 2021-06-26 by fergu