ivanseidel/DueTimer #68: DueTImer and serialUSB

DavidTol, 2019-01-17:
Hi,
I have a problem with DueTimer: if I have an active timer the seriaUSB.readbytes() function freeze my Arduino DUE. I am not an expert on Arduino but I think there are a conflict inside theese two library (same timer?).

Thanks for help

BR,
David

MicahStevens, 2019-04-04:
I'm seeing a conflict with the delay() function. Might be related? This was working well - maybe recent Arduino updates has caused a problem. I'm still digging into it.

ivanseidel, 2019-04-04:
Try doing a delay with micros(), like: `while(micros() - start < DELAY);`

It might work, since the delay function depends on millis, and when you are
inside a timer callback, that timer stops counting, so it's like it's
froozen in time, that's why you get stuck.

However, micros() is different and doesn't depends on a timer

MicahStevens, 2019-04-05:
Hi - yeah, I deleted my comment when I realized I was inside a timer interrupt when I was calling delay() - not a good idea. It was just debugging code anyway. Now it's working fine. 

To @DavidTol - I'm using this library with SerialUSB currently and it seems stable for me, if that helps. I haven't had any problems other than my silly delay() mistake.