ivanseidel/DueTimer #72: Change timer value on the fly

2kohm, 2019-11-30:
Hello,
i want to change the timer intervall on the fly

i generate the intervall time with an formula

```
int myVar = 0;
double ticks = 60000000.0 / (myVar * 96.0);

void clock1()
    {
        // do stuff here
    }

void setup()
{
	Timer2.attachInterrupt(clock1); 
}

void loop () 
{
    if ( something == 1) 
    {
        Timer2.start(ticks);
    }

    if ( somethingElse)  
        {
            myVar = 120;
        }
    }
}
```
ho can i update the changed value on the fly? do i have to stop and start the Timer again ?

budryerson, 2021-10-07:
Count me in!
I start with this framework:
```
void T3_Handler()
{
    // do something
}

void setup()
{
    Timer3.attachInterrupt( T3_Handler);
    Timer3.start( 500000);   // timer interrupts every 0.5 seconds
}

void loop()
{
    delay( 2000);                  // do something 4 times
    Timer3.setFrequency( 100000);  // set timer to interrupt every 0.1 seconds
    delay( 2000);                  // do something 20 times
    Timer3.setFrequency( 500000);  // set timer back to 0.5 seconds
}
```
That does not work.  In fact, a sketch like this will just seize up like a cheap watch.
So I tried:
`Timer3.setFrequency(100000).start()`
And I tried:
`Timer3.setFrequency(100000); Timer3.start()`
And I tried:
`Timer3.start(100000);`
None of it works!
I want to accelerate a stepper motor.
a) What is going wrong; and
b) What am I supposed to do?

Shoxx98, 2022-11-11:
+1
got it working with .setfrequency, then doing .start() but this seems rather inefficient.

ivanseidel, 2022-11-11:
Maybe `setFrequency` could call  `.start` at the end automatically. And, if a frequency `<= 0` was assigned to the parameter, `.stop` would be called.

Can anyone test if calling `.start` works after all?

iddq, 2022-11-14:
setFrequency calls TC_Configure which disables clock and interrupts. Although interrupts are re-enabled by setFrequency later, the clock is still disabled which can be enabled by TC_Start.

Shoxx98, 2022-12-16:
has anyone figured out how to stop the disabling of the clock?
I am messing around with it rn and simply commenting out  `pTcCh->TC_CCR = TC_CCR_CLKDIS ;` in TC_Configure makes it stop entirely

iddq, 2022-12-17:
It is not possible.

iddq, 2022-12-17:
please explain how did you recompile the sam core after the modification of tc.c?