ivanseidel/DueTimer #77: Multiple Timers Hanging

RRDajay, 2020-11-26:
Hi, I've modified the example code from the multiple timer folder with my desired interrupt time (1000 microseconds for Timer0 and 5000 microseconds for Timer3). From the arduino serial monitor, It seems that the arduino hangs at 5000us. Does this mean that the arduino does not know what ISR to prioritize? I've attached the code and the output of the arduino serial monitor for visualization. 

```
void firstHandler(){
	Serial.println("[-  ] First Handler!");
}

void secondHandler(){
	Serial.println("[ - ] Second Handler!");
}

void thirdHandler(){
	Serial.println("[  -] Third Handler!");
}

void setup(){
	Serial.begin(115200);

	Timer0.attachInterrupt(firstHandler).start(1000);
	Timer3.attachInterrupt(secondHandler).start(5000);

}

void loop() {

}
```

[image]

Am I missing something? Its my first time using the library. Any help would be greatly appreciated. Thank you.

kapyaar, 2021-03-17:
Can you try detachInterrupt and reenable interrupt in the ISR, and see if that keeps it going?

Eagle735, 2021-04-22:
I have the same Problem. DetachInterrupt and reenable it doesnt work.

iddq, 2022-12-19:
Don't use print in ISR. Please close this issue as it is invalid.

TheOnlyBeardedBeast, 2024-08-17:
I have the same issue, I can not run 2 timers at the same time. At my program I start a timer, which runs periodically, the second timer is not running periodically, I change its frequency, and start it again. Somehow my second timer works all the time, but the first timer doe not, I made sure they are using different hardware timers;